OnionIoT/Onion-Docs #3: Some other languages examples ?

jorismulliez, 2016-12-16:
Hi,

I'm glad to see there is a new documentation for the omega2 !
But I can't find any example for php, ruby, nodejs ...

How do we configure the omega2 to execute one of this languages and how can we use the GPIO through this languages ?

thanks for your help,

cozy2963, 2017-01-19:
I had been searching for some other info as well! Do you guys plan to roll out any more tutorials?

Psype, 2017-02-04:
Guys, Omega2 is based on LEDE/OpenWrt. Why not trying this ?
https://wiki.openwrt.org/doc/techref/opkg 
;)

KitBishop, 2017-02-04:
@Psype That reference is certainly useful for understanding usage of opkg, and similar documentation for LEDE can be found at https://lede-project.org/docs/user-guide/opkg
And in general, there is a lot of interesting stuff to be found on OpenWrt and LEDE at (respectively): https://wiki.openwrt.org/doc/start and https://lede-project.org/docs/start
However, the original question was about tutorials on (e.g.) php, ruby, nodejs etc.  Documentation on these and other languages is not likely to be found under OpenWrt and LEDE since these languages are not specific to these operating systems alone but rather to a wide range of related Linux (and other) based operating systems.
Using opkg you can ascertain whether the language packages is available for installation, but for specific usage of a language, one is better to search for general documentation on that language