OnionIoT/Onion-Docs #18: 1-wire tutoriel for omega 2?

jorismulliez, 2017-01-22:
Can you provide a tutorial for using 1-wire on omega2/2+ ?
https://wiki.onion.io/Tutorials/Reading-1Wire-Sensor-Data

Thanks,

pieman64, 2017-01-22:
@jorismulliez what errors did you get when you tried to use 1Wire with the 2 and 2+?

jorismulliez, 2017-01-23:
I've not try yet. Because some other users got errors :
https://community.onion.io/topic/1310/1-wire-temperature-sensor-ds18b20-on-omega2/7

jorismulliez, 2017-01-28:
Hi,

Any news for this problem ?

Thanks,

greenbreakfast, 2017-02-01:
Planning to work look into it this week. Have you tried it out on your end?

jorismulliez, 2017-02-01:
Hi,

I didn't try yet because I'm scared of burn my omega with a mis-configured pin!
And because I read on the thread pointed before that 1-wire isn't functional.

Thanks for your help!

pieman64, 2017-02-02:
@greenbreakfast looking forward to the fix Lazar.

pieman64, 2017-02-03:
@greenbreakfast thanks for the b151 firmware upgrade.

@jorismulliez confirmed as working on my 2+

[image]

greenbreakfast, 2017-02-04:
@pieman64 no problem! I'll update the known issues page.
Some other issues popped up in the firmware so I haven't advertised it yet :)

greenbreakfast, 2017-02-04:
Whoops, guess we still need a tutorial, gonna reopen this issue

greenbreakfast, 2017-02-05:
Added an article! Find it published here: https://docs.onion.io/omega2-docs/communicating-with-1w-devices.html

closed 2017-02-05 by greenbreakfast

jorismulliez, 2017-02-05:
@greenbreakfast Thanks for the work,
But it seems not to work for me, see : https://community.onion.io/topic/1310/1-wire-temperature-sensor-ds18b20-on-omega2/32