OnionIoT/Onion-Docs #25: Using-GPIOs.md Inaccurate

pjobson, 2017-02-19:
Using-GPIOs.md shows that you can use `gpioctl` to control say a LED.

"You can configure your GPIO pin to supply power to a load with your Omega, for example, if you were powering an LED."

    root@Omega-ADBD:~# gpioctl dirout 6
    Using gpio pin 6.
    root@Omega-ADBD:~# gpioctl dirout-high 6
    Using gpio pin 6.
    root@Omega-ADBD:~# gpioctl get 6
    Using gpio pin 6.
    Pin 6 is HIGH
    root@Omega-ADBD:~# gpioctl dirout-low 6
    Using gpio pin 6.
    root@Omega-ADBD:~# gpioctl get 6
    Using gpio pin 6.
    Pin 6 is LOW

Nothing happens, I was expecting the LED to do something... though the documentation isn't clear as to what it is supposed to do.  I'm guessing turn on and off.

greenbreakfast, 2017-02-20:
As mentioned in the section on [Multiplexed GPIOs](https://docs.onion.io/omega2-docs/using-gpios.html#using-gpios-multiplexing), GPIO6 is set to SPI_CS1 by default. 
Try using GPIO0 or GPIO1

pjobson, 2017-04-02:
gpio pin 1 works.

closed 2017-04-02 by pjobson