OnionIoT/Onion-Docs #26: Expansion-Dock.md Inaccurate & expled segfaults

pjobson, 2017-02-19:
According to Expansion-Dock.md you can use `expled` to change the LED color on the board.  You can't actually do anything with it.

Various tries, for all the LED remained blue.  If you put a 00 as a value the program segfaults.

    root@Omega-ADBD:~# expled 0xf21133
    Setting LEDs to: f21133
    Duty: 6 94 80
    root@Omega-ADBD:~# expled 0xff1133
    Setting LEDs to: ff1133
    Duty: 0 94 80
    root@Omega-ADBD:~# expled 0xff0000
    Setting LEDs to: ff0000
    Duty: 0 100 100
    Segmentation fault
    Segmentation fault
    root@Omega-ADBD:~# expled 0x00ff00
    Setting LEDs to: 00ff00
    Duty: 100 0 100
    Segmentation fault
    Segmentation fault
    root@Omega-ADBD:~# expled 0x0000ff
    Setting LEDs to: 0000ff
    Duty: 100 100 0
    Segmentation fault
    Segmentation fault

greenbreakfast, 2017-02-20:
Yep, `expled` currently doesn't work since it relies on fast-gpio, this is listed as one of the known issues in the firmware: https://docs.onion.io/omega2-docs/known-firmware-issues.html

greenbreakfast, 2017-03-13:
This is now fixed, please update your firmware

(fix is reflected at: https://docs.onion.io/omega2-docs/known-firmware-issues.html)

closed 2017-03-13 by greenbreakfast