OnionIoT/Onion-Docs #34: Cannot connect to Onion Firmware Server

canexpat, 2017-04-16:
Attempting to use oupgrade but running into problems.  Help?  Here is a snipping of the error message.

[image]

greenbreakfast, 2017-05-15:
Can you try this method: https://docs.onion.io/omega2-docs/manual-firmware-installation.html

danielsitek, 2017-07-20:
@canexpat try manual firmware update via USB drive and then run `$ sysupgrade  -n`.

I had a problem with connecting my Omega to the router and this finally worked for me.

More info:
* https://docs.onion.io/omega2-docs/manual-firmware-installation.html
* https://docs.onion.io/omega2-docs/usb-storage.html