OnionIoT/Onion-Docs #42: syntax error: unexpected "(" (expecting ")")

viwell, 2017-06-14:
Hi, I'm trying to get data of a DHT22 sensor in Omega2, but am getting the following error:

```bash
line 1: syntax error: unexpected "(" (expecting ")")
```

Any ideas about it? Thanks.

greenbreakfast, 2017-06-15:
hi, which guide are you using?

viwell, 2017-06-15:
This one https://github.com/OnionIoT/Onion-Docs/blob/master/Omega2/Projects/PHP-DHT11-DHT22-Sensor-Examples.md

viwell, 2017-06-15:
Hi, seems that checkHumidity binaries you guys provided are for the Omega1, where can I find binaries for the Omega2?

DanOBrien, 2017-06-19:
I'm getting the same message:

root@onion1:~/checkHumidity/bin# ./checkHumidity 1 DHT11
./checkHumidity: line 1: syntax error: unexpected "(" (expecting ")")

I'm using this hardware:
Codename:	Ando
Model:	Onion Omega2+
Revision:	R1
Firmware:	0.1.10 (b160)

The page: https://wiki.onion.io/Tutorials/PHP-DHT11-DHT22-Sensor-Examples  and  https://github.com/OnionIoT/Onion-Docs/blob/master/Omega2/Projects/PHP-DHT11-DHT22-Sensor-Examples.md seem to be the same thing, but not valid for Omega2.

mdoary, 2017-07-07:
Anyone get this working?

h0l0gram, 2017-08-04:
@DanOBrien @viwell 
For the Omega2(+) adapted sources see https://github.com/h0l0gram/omega2-checkHumidity.
There is even a compiled binary for **firmware 0.1.10 b160** available.

greenbreakfast, 2018-06-01:
We've rolled the DHT sensor code into our package repo, for more details see: https://onion.io/2bt-reading-dht-sensor-data/

closed 2018-06-01 by greenbreakfast