OnionIoT/Onion-Docs #44: Getting "Error compiling for board Onion Arduino Dock"

rodrigocmendes, 2017-07-20:
Followed all the instructions, but when I try to upload the code to the arduino dock, I'm receiving this from Arduino IDE:

"C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc-ar" rcs  "C:\Users\rodri\AppData\Local\Temp\arduino_build_66107\core\core.a" "C:\Users\rodri\AppData\Local\Temp\arduino_build_66107\core\wiring_pulse.S.o"
Error running c:\users\rodri\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: CreateProcess

Using library Adafruit_Sensor-master at version 1.0.2 in folder: C:\Users\rodri\Documents\Arduino\libraries\Adafruit_Sensor-master 
Using library DHT-sensor-library-master at version 1.3.0 in folder: C:\Users\rodri\Documents\Arduino\libraries\DHT-sensor-library-master 
exit status 1
Error compiling for board Onion Arduino Dock.

gbo-actual, 2017-07-20:
What version of the Arduino IDE are you running, and do you have verbose output enabled on compilation in the settings? If not, can you post the error again after enabling it.

rodrigocmendes, 2017-07-20:
Hi, I'm using the updated version 1.8.3.

Yes, I'm with the verbose output enabled.
I'm posting again the full report with the error at the end. See below:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\rodri\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\rodri\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\rodri\Documents\Arduino\libraries -fqbn=onion:avr:arduinodock -ide-version=10803 -build-path C:\Users\rodri\AppData\Local\Temp\arduino_build_66107 -warnings=none -build-cache C:\Users\rodri\AppData\Local\Temp\arduino_cache_380066 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\rodri\Documents\Arduino\libraries\Onion_Arduino_Library\examples\weatherStation\weatherStation.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\rodri\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\rodri\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\rodri\Documents\Arduino\libraries -fqbn=onion:avr:arduinodock -ide-version=10803 -build-path C:\Users\rodri\AppData\Local\Temp\arduino_build_66107 -warnings=none -build-cache C:\Users\rodri\AppData\Local\Temp\arduino_cache_380066 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\rodri\Documents\Arduino\libraries\Onion_Arduino_Library\examples\weatherStation\weatherStation.ino
Using board 'arduinodock' from platform in folder: C:\Users\rodri\AppData\Local\Arduino15\packages\onion\hardware\avr\1.0.0
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\rodri\AppData\Local\Temp\arduino_build_66107\sketch\weatherStation.ino.cpp" -o "nul"
"C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Users\rodri\Documents\Arduino\libraries\Adafruit_Sensor-master" "C:\Users\rodri\AppData\Local\Temp\arduino_build_66107\sketch\weatherStation.ino.cpp" -o "nul"
"C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Users\rodri\Documents\Arduino\libraries\Adafruit_Sensor-master" "-IC:\Users\rodri\Documents\Arduino\libraries\DHT-sensor-library-master" "C:\Users\rodri\AppData\Local\Temp\arduino_build_66107\sketch\weatherStation.ino.cpp" -o "nul"
"C:\Users\rodri\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Users\rodri\Documents\Arduino\libraries\Adafruit_Sensor-master" "-IC:\Users\rodri\Documents\Arduino\libraries\DHT-sensor-library-master" "C:\Users\rodri\Documents\Arduino\libraries\DHT-sensor-library-master\DHT.cpp" -o "nul"
[... cut ...]
Error running c:\users\rodri\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: CreateProcess

Using library Adafruit_Sensor-master at version 1.0.2 in folder: C:\Users\rodri\Documents\Arduino\libraries\Adafruit_Sensor-master 
Using library DHT-sensor-library-master at version 1.3.0 in folder: C:\Users\rodri\Documents\Arduino\libraries\DHT-sensor-library-master 
exit status 1
Error compiling for board Onion Arduino Dock.

greenbreakfast, 2017-09-19:
Looks like your error is here:
```
Error running c:\users\rodri\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\bin../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: CreateProcess
```

This doesn't look to be Arduino Dock related. Let's confirm that, please try compiling for an Arduino Uno board, if the error again then it has something to do with your Arduino IDE installation.

rodrigocmendes, 2017-09-19:
Hi, I've found out what was the problem: Windows 10.
You cannot downlod the arduino installer from Arduino site and install by  yourself.
You have to download it from Windows Store. Just it.

Thanks @greenbreakfast !!!

closed 2017-09-19 by rodrigocmendes

greenbreakfast, 2017-09-22:
No problem! Glad to hear it worked out!