OnionIoT/Onion-Docs #47: Omega2 GPIO7,8,9 can't actually be used for general purpose I/O

lis0r, 2017-08-15:
Whilst GPIOs 7, 8, and 9 might technically be GPIOs on the SoC, they can't be used as such on the module, due to the flash using their SPI channel. This is alluded to in a dark corner of the written documentation, but could be more clearly communicated by not marking them as GPIOs on the pinout diagram.

greenbreakfast, 2018-06-01:
updated the diagram! will update the site as well

PythonSmith, 2020-01-22:
All documents need to be updated, i tried https://docs.onion.io/omega2-starter-kit/starter-kit-temp-sensor.html but i do not have GPIO 19 on my dock and tried GPIO 9 and 8. GPIO 0 is working though

greenbreakfast, 2020-01-22:
what could be updated in the 1-Wire Temperature Sensor article to make it clearer?

PythonSmith, 2020-01-25:
Oh, i forgot that i have a power dock and not a dock expansion
We should note at the "Building the circuit" that basically all GPIOs can be used expect 7, 8 and 9. And tell the reader that he only needs to change the GPIO in the code too.

I would also find it cool to have some document which describes how we can have multiple w1-buses (probably overkill for a Onion but it maybe good to know)

greenbreakfast, 2020-01-29:
updated the document to make it clear that other GPIOs can be used, and pointed to our article on [using the gpios](http://docs.onion.io/omega2-docs/using-gpios.html#using-gpios-multiplexing)

The article on the docs.onion.io website will be updated automatically in a few hours.

Btw, it's definitely doable to instantiate multiple buses. Just need to instantiate them with unique GPIOs.
See this article for more details on instantiating one-wire buses: https://onion.io/2bt-reading-temperature-from-a-1-wire-sensor/
It goes into more detail about the instantiation commands, while the starter kit article puts all of that in a Python program.

closed 2020-01-29 by greenbreakfast