OnionIoT/Onion-Docs #64: basics in Project Book

alysdexia, 2018-03-25:
Find function is completely broken.
it's -> its, retard
" - ", "- " -> —, : 
will -> shall
comma splices
colon splices
number splice
"monocrome"
Project Book says / is at bottom of a tree but then refers to other directories as below and rootward as up.
"gets their"
fast[er] -> soon[er], short[li][er], swift[li][er]
command like -> command line
"Resisters"
resorts to incorrect conventional polarity
mistakes behind for above
"little bit"
thin -> fine, slim
super useful -> superuseful
method -> methods
or less -> or fewer
data that's -> data that're
URLs, images, and code off the edge on pages 80, 81, 82, 103, 104, 111, 128, 129, 146, 152, 213, 219, 247, 256, 259, 272, 273, 284, 302
"create quickly create"
nice < niais < nescius := not-skilled -> well
[that][This] data -> [those][These] data
[to] [we] setup -> [to] [we] set up
create -> to create
Length -> Height
data is -> data are
out script -> our script
"periods of time"
away from -> frowards
length -> span
duration -> length
[left][right] most -> [left][right]most
Yyu -> You
apart -> apart.
Onion' -> Onion's
blank page 255
table overlap on page 263
corrupt image on page 274
Andriod -> Android
widow on page 275
large square piece -> block
"storageL"
mislink on page 309, 319
"on connected"
be be -> to be
routers -> router's
is, the -> is the
travel < travail < trepale -> go, fare, wend
deadlink on page 322

greenbreakfast, 2018-03-27:
Hi, while we appreciate you pointing out typos, we do not tolerate name calling. Especially when it comes to hundreds of pages of content provided free of charge.
If you would like to be helpful, refrain from name-calling and give an indication in which each project can be found.

alysdexia, 2018-03-27:
týpo = stroke; dýstýpo = misstroke.  It doesn't matter if the content is free or not; the frequent consistent misspellings of the same word are that annoying; that you don't accept impressions (not "name calling", as "greenbreakfast" is a name) of how a reader finds the content rather than whether the comment is dishonest or wrong tells you don't care how bad the content is.

Most of these mistakes are in the same book, mentioned near the top.  The layout looks fine in HTML but not PDF.  Most of the mistakes can be gotten with Find; others are widespread.  Nonetheless they're listed in order.

guybli, 2018-03-27:
> it's -> its, **retard**  .

 this is what greenbreakfast is talking about. your grammatical improvements are appreciated. but calling people retard is much more annoying than bad English.

also, since its such an annoyance for you, Github does give you the option to fork the projects, apply the changes you would like to see and send a pull request. you don't need to highlight what is wrong here; you have the power to fix it yourself. (this is what open source collaboration is all about.)

alysdexia, 2018-03-28:
Dictional improvements you mean, besides compositional.  Insults may be annoying to the guilty but so are good rules.  You broke one oppositely: its -> it's.

GitHub can't correct the PDF which nobody proofreads, and I can't edit PDFs, especially readonlies.  It'd be easier if all of the docs were concatenated like the PDF along with the latter's page numbers.

greenbreakfast, 2018-03-28:
Like I said, we welcome and encourage suggestions, improvements, typo fixes, but this is not a place for insults and it will not be tolerated. 

The PDF is made from this repo, all of the source text for the projects can be found here: https://github.com/OnionIoT/Onion-Docs/tree/master/Omega2/Projects