OnionIoT/Onion-Docs #68: GCC is no longer available in package repositories

nicolapalavecino, 2018-04-30:
https://docs.onion.io/omega2-docs/c-compiler-on-omega.html says that if two lines are uncommented in the file `/etc/opkg/distfeeds.conf` you can add some package repositories for you to be able to compile C and C++ programs in the Omega. I found out that the gcc compilers are not available in any repository. The following error message appears when I try to install gcc with `opkg install gcc`:

```
Unknown package 'gcc'.
Collected errors:
 * opkg_install_cmd: Cannot install package gcc.
```

greenbreakfast, 2018-05-01:
Hmm looks like the LEDE people took it out of their repo. We'll look into it and get back to you!

corndog2000, 2018-05-19:
Please fix! The Onion Omega2+ is pretty much useless to me without C++ and C.

nicolapalavecino, 2018-05-19:
@corndog2000 you can still cross-compile C and C++ programs in a linux environment. You can see [c-cross-compile-example](https://github.com/OnionIoT/c-cross-compile-example) to know how to do it.

But I agree with you, Omega2+ is useless if you can't compile within the Omega OS. @greenbreakfast could you see what happened?

corndog2000, 2018-05-19:
I would do that except the LEDE Build System always fails to build.