OnionIoT/Onion-Docs #70: Cannot setup build system

emptyHeap, 2018-05-04:
Hello, dont really know is it Onion-Docs, OnionIoT issue or I just missing something simple. Section Cross-Compiling->Build System Root describe setup build system process, but it `make -j20` command finished with following error:

```
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/konstantin/workplace/onionIoT/source/include/toplevel.mk:198: recipe for target 'world' failed
make: *** [world] Error 1
```

re-runing make as `make -j1 V=s` have tons of output, but it ends with following error:

```
make[6]: Entering directory '/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3/debugfs'
	CC debug_cmds.c
	CC extent_cmds.c
	CC create_inode.o
./../misc/create_inode.c: In function 'set_inode_xattr':
./../misc/create_inode.c:136:9: warning: implicit declaration of function 'llistxattr'; did you mean 'lstat64'? [-Wimplicit-function-declaration]
  size = llistxattr(filename, NULL, 0);
         ^~~~~~~~~~
         lstat64
./../misc/create_inode.c:172:16: warning: implicit declaration of function 'lgetxattr'; did you mean 'getdate_r'? [-Wimplicit-function-declaration]
   value_size = lgetxattr(filename, name, NULL, 0);
                ^~~~~~~~~
                getdate_r
./../misc/create_inode.c: At top level:
./../misc/create_inode.c:395:18: error: conflicting types for 'copy_file_range'
 static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
                  ^~~~~~~~~~~~~~~
In file included from ./../misc/create_inode.c:19:0:
/usr/include/unistd.h:1110:9: note: previous declaration of 'copy_file_range' was here
 ssize_t copy_file_range (int __infd, __off64_t *__pinoff,
         ^~~~~~~~~~~~~~~
Makefile:417: recipe for target 'create_inode.o' failed
make[6]: *** [create_inode.o] Error 1
make[6]: Leaving directory '/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3/debugfs'
Makefile:404: recipe for target 'all-progs-recursive' failed
make[5]: *** [all-progs-recursive] Error 1
make[5]: Leaving directory '/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3'
Makefile:339: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3'
Makefile:49: recipe for target '/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3/.built' failed
make[3]: *** [/home/konstantin/workplace/onionIoT/source/build_dir/host/e2fsprogs-1.43.3/.built] Error 2
make[3]: Leaving directory '/home/konstantin/workplace/onionIoT/source/tools/e2fsprogs'
tools/Makefile:147: recipe for target 'tools/e2fsprogs/compile' failed
make[2]: *** [tools/e2fsprogs/compile] Error 2
make[2]: Leaving directory '/home/konstantin/workplace/onionIoT/source'
tools/Makefile:145: recipe for target '/home/konstantin/workplace/onionIoT/source/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.tools_install_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynnny' failed
make[1]: *** [/home/konstantin/workplace/onionIoT/source/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.tools_install_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynnny] Error 2
make[1]: Leaving directory '/home/konstantin/workplace/onionIoT/source'
/home/konstantin/workplace/onionIoT/source/include/toplevel.mk:198: recipe for target 'world' failed
make: *** [world] Error 2
```

I am not really familiar with cross compiling and stuck here.

ranabasheer, 2018-09-13:
Was this issue fixed? I am still having the above problem by just running make

greenbreakfast, 2018-09-17:
Are you compiling using our Docker image or on your own system? 
We have a [few notes about both options](https://github.com/OnionIoT/source#using-this-build-system) in our LEDE source repo. Note that we recommend using the Docker image, there's [in-depth instructions](https://onion.io/2bt-cross-compiling-c-programs-part-1/) for that available as well.

dudanov, 2018-10-24:
https://github.com/openwrt/openwrt/pull/781

dudanov, 2018-10-24:
uiuygggggggfggtg+++++++++

dudanov, 2018-10-24:
This issue described and closed [here](https://github.com/openwrt/openwrt/pull/781).