OnionIoT/Onion-Docs #71: Cannot install any kernel modules

corndog2000, 2018-06-05:
The documentation on OPKG says:

> Note that your Omega cannot install kernel modules from the default LEDE repos. opkg only allows installation of kernel modules compiled by the same machine that compiled the firmware. This is to ensure that all installed kernel modules are the exact same version as the device’s kernel. It would be pretty disastrous to install a module that expects a different version of the overall kernel so opkg never lets it happen!

My Onion Omega 2+ has firmware version 0.2.0 (b186), which uses the 4.4.74-1 kernel. Whenever I try to install any kernel modules the install fails and returns the error

`Collected errors:
 satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-hfs:
      kernel (= 4.4.135-1-fd6e4535420b1cebe485c935e0d20c0d) *
 opkg_install_cmd: Cannot install package kmod-fs-hfs.
 pkg_run_script: package "kmod-input-joydev" postinst script returned status 255.
 opkg_configure: kmod-input-joydev.postinst returned 255.
`

I cannot install any kernel modules except for kmod-input-core but I had to use the --nodeps install flag.
`opkg install kmod-input-core --nodeps
`

greenbreakfast, 2018-06-05:
Yep, as the documentation mentions, it is only possible to install kernel modules from the Onion Package Repo. It is **not possible** to install kernel modules from any **other repos** because of the reasons outlined in the text you've quoted. 

How are you trying to install these kernel modules?

corndog2000, 2018-06-05:
With opkg like
`opkg install kmod-fs-hfs`
in the tutorial

That command returns

Installing kmod-fs-hfs (4.4.135-1) to root...
Downloading http://downloads.lede-project.org/releases/17.01-SNAPSHOT/targets/ramips/mt7688/packages/kmod-fs-hfs_4.4.135-1_mipsel_24kc.ipk
Configuring kmod-input-joydev.
failed to find a module named joydev
Collected errors:
  satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-hfs:
       kernel (= 4.4.135-1-fd6e4535420b1cebe485c935e0d20c0d) 
  opkg_install_cmd: Cannot install package kmod-fs-hfs.
  pkg_run_script: package "kmod-input-joydev" postinst script returned status 255.
  opkg_configure: kmod-input-joydev.postinst returned 255.

greenbreakfast, 2018-06-05:
Judging by the URL it's downloading from, it looks like you have the **LEDE package repos** enabled. What's happening is that it's trying to install the kernel modules with the higher verison number and it happens that the LEDE package repo has the kernel modules built for a later version of the kernel. 
 
You'll have to disable the LEDE package repos so that opkg will only try to install kernel modules from the Onion package repos. For more info on enabling/disabling package repos, see this Docs article: https://docs.onion.io/omega2-docs/using-opkg.html#using-opkg-switch-to-lede-repos