OnionIoT/Onion-Docs #87: How to convert Onion-Docs to a offline documentation

mickpf, 2021-07-20:
Dear Sirs,

Can you please provide me with the instructions on how to create offline documentation from "Onion-Docs".

Thanks in advance,
Michael

crispyoz, 2021-07-20:
Michael if you just want the PDF version of the docs these are available at 

https://github.com/OnionIoT/Onion-Docs/tree/master/Omega2/Documentation

If you are asking how to create docs from the repo files you need to understand how to use BookDown, the repo is here:

https://github.com/rstudio/bookdown

mickpf, 2021-07-20:
Hello Crispyoz,

In fact, I'm looking for a PDF version of the documentation because I'm not always connected to the internet.
Please forgive me for the sarcasm: Maybe I'm blind, but there are no PDF files in the directory you specified (in subdirs also).

crispyoz, 2021-07-20:
Sorry yes you are correct the pdfs are not there, look at https://docs.onion.io/omega2-docs/

crispyoz, 2021-07-20:
For more support and advice, join the the Onion  Omega community https://community.onion.io/

mickpf, 2021-07-20:
There is no PDF version available there either.
That's why I asked how to convert "Onion Docs" to PDF.

That it ends like almost everywhere, because you don't want to bother, I already thought.
Thanks for nothing...

closed 2021-07-20 by mickpf

greenbreakfast, 2021-07-21:
For reference, our CI system uses the markdown files in this repo and bookdown to build the html documentation at: http://docs.onion.io/
We have no process for building PDF documentation - so far online documentation has served our users well.

If you need offline PDF documentation, I would suggest taking matters into your own hands. You can try scraping the individual pages at http://docs.onion.io/ and using a [utility to convert the html into PDFs](https://www.npmjs.com/package/html-pdf-node).